devicehive/esp8266-firmware #1: хотелось бы видеть все это в переводе на родной русский язык

edgiru, 2015-08-17:
автор помоему Русский или я ошибаюсь?
да и еще кое что на DeviceHive не возможно зарегистрироваться во всех предложенных вариантах выходят ошибки при регистрации.

Nikolay-Kha, 2015-08-17:
На русском, как этим всем пользоваться, будет опубликовано на гиктаймсе. Первая часть - http://geektimes.ru/company/dataart/blog/259274/ - она о том что из оборудования вообще нужно иметь. Вторая часть выйдет совсем скоро она и будет посвещена процессу прошивки и конфигурирования.
А на сайте DeviceHive попробуйте использовать другой браузер.

edgiru, 2015-08-17:
Спасибо что ответили Николай. Прошивку уже прошил все ок, но вот не пойму что дальше делать. Правильно ли я понял что надо поднимать свой сервер DeviceHive который только на .NET и Java? Другой браузер тоже не помогает. при любом входе github, google, facebook одна и таже ошибка, скрин: https://goo.gl/kFlr7f
хотел зарегистрироваться просто как обычно имя пользователя, емайл набрал пишет что такой емайл зарегистрирован попробуйте сбросить пароль (хотя я не регистрировался еще). Хорошо поппытался сбросить пароль ссылка пришла минут через 20. нажимаю ссылку в письме перекидывает меня на сайт  DeviceHive жму сбросить пароль, набираю новый пароль выходит ошибка "наберите старый пароль" оп и тут засада )). вообщем в плейграунд так и не попал поиграться.

Wanderernk, 2015-08-17:
Доброго времени суток.
Та же проблема с регистрацией в песочнице (Playground)

Nikolay-Kha, 2015-08-18:
Пробуйте сейчас, по идее починили.

Сервер существует как для .NET так на Java. Однако его можно развернуть в одну команду при помощи Docker ( https://ru.wikipedia.org/wiki/Docker ):
docker run -d --name=devicehive -p 8080:80 -p 9001:9001 devicehive/devicehive-standalone
Полное описание на страничке Docker’а: https://registry.hub.docker.com/u/devicehive/devicehive-standalone/

edgiru, 2015-08-18:
Огромное спасибо Николай! все заработало. буду вечером эксперементировать.

Nikolay-Kha, 2015-08-18:
статья на русском вышла - http://geektimes.ru/company/dataart/blog/260460/

closed 2015-09-01 by edgiru