devicehive/esp8266-firmware #2: dht22

edgiru, 2015-08-18:
Добрый вечер
хотелось бы увидеть пример с использованием датчика температуры и влажности dht22.

p.s. с bmp180 уже разобрался. нигде не написано может кому то пригодится: SDA:GPIO0 и SCL:GPIO2 пример: bmp180-i2c.html сразу заработал.

Nikolay-Kha, 2015-08-18:
ну по самому семплу да, не так очевидно, но есть дока в которой:
https://github.com/devicehive/esp8266-firmware/blob/master/DeviceHiveESP8266.pdf
страница 16 и 17
"SDA" ­ GPIO port number for SDA data line. If not specified, previous pins will be used. Default
is "0".
"SCL" ­ GPIO port number for SCL data line. If not specified, previous pins will be used. Default
is "2".
Можно перемапить на любые другие ноги если послать хотя бы одну команду указав в параметрах SDA и SCL с другими пинами.

dht22 подключается по шине 1-wire, которая в текущем релизе еще не релизована. Но скоро будет - можно будет iButton, DS18B20 и тот же DHT22 подключать.

edgiru, 2015-09-06:
Николай а релиз 0.3v не скомпилируете? а то я пока в этих делах не очень разбираюсь))

Nikolay-Kha, 2015-09-07:
Бинарные образы:
https://github.com/devicehive/esp8266-firmware/releases
Пример для DHT https://github.com/devicehive/esp8266-firmware/blob/master/examples/dht-onewire.html
Он же на видео: https://www.youtube.com/watch?v=IuvxwCPNZCc
В руках был только DHT11, поэтому сильно извиняюсь если вдруг (ошибиться там реально сложно) для DHT22 неправильно единицы переводятся.

edgiru, 2015-09-07:
не совсем понято как работает. на gpio0 висит и bmp и dht? как это может быть? а что бы GPIO поменять это надо в прошивке в исходниках поменять или заново компилировать?

Nikolay-Kha, 2015-09-07:
Для I2C и onewire можно выбрать любые пины чипа. Саму прошивку при этом трогать не надо. Достаточно в исходниках примера https://github.com/devicehive/esp8266-firmware/blob/master/examples/dht-onewire.html поменять на строке 23:
var PIN = 0;
на нужный номер вывода. В примере bmp180 SDA и SCL так же можно перемапить на любые другие выводы. Только UART и SPI имеет апаратные выводы на чипе, которые нельзя изменить.

closed 2015-10-11 by Nikolay-Kha