devicehive/esp8266-firmware #3: Не работает после прошивки

alex-s, 2015-08-24:
Здравствуйте, вероятно дело не в вашей прошивке, но может вы сможете мне что-нибудь посоветовать. Подключаю все контакты, запускаю флешер, все ок, выдергиваю контакт GPIO0, запускаю еще раз, пишет "Can not detect port. Check if device connected and driver is installed.", и, не знаю важно ли, на модуле не горят диоды, хотя в режиме прошивки горит красный.

Система OS X 10.10.5, esp8266-01, USBtoTTL CH340G.

Заранее спасибо!

Nikolay-Kha, 2015-08-25:
Здравствуйте!
Красный светодиод на плате именуемой esp-01 подключен напрямую к питанию, т.е. он горит всегда когда на плате есть питание и не важно, что там происходит с процессором в это время.  Если питание есть, он должен гореть. У вас же не горит. Возможно, отсутствует контакт с землей и питание через защитный диоды на чипе проходит через GPIO0, проверьте провода подключающие землю(он же минус, он же GND). Также убедитесь, что Вы отключаете  GPIO0, а не землю. Распиновку можно найти здесь - https://github.com/devicehive/esp8266-firmware/raw/master/DeviceHiveESP8266.pdf страница 4, нижний рисунок.

edgiru, 2015-08-25:
точно не в прошивке. Я пробовал прошивку очень качественная, без глюков. Мне очень часто попадались глючные модули. Китайцы их очень плохо пропаивают. Согласен с Николаем. Если у Вас отваливается "питание" во время прошивки вы можите "убить" свой модуль. Делайте хороший контакт без "дребезга".

alex-s, 2015-08-25:
Спасибо за ответы, я еще раз все переповерю, и если все попрежнему, попробую найти другой модуль, может действительно китайцы перемудрили.

closed 2015-08-25 by alex-s