devicehive/esp8266-firmware #4: примеры

edgiru, 2015-09-08:
Я пытаюсь разобраться. т.к. я не программист а системщик пытаюсь начать с самого простого. Немного изменил Ваш пример. Но не работает висит на "Connecting... Refresh page for canceling." пробовал вручную командами в личном кабинете на devicehive все ок работает dht22 дает ответ.

подправил только тут:

  </head>
    <body onLoad="initform(this.form);">  
    <form name="form">
      <input type="text" name="url" style="color: #777;" value="DeviceHive API Url" onfocus="if (this.value == 'DeviceHive API Url') {this.value = ''; this.style.color = '#000';}" onblur="if (this.value == '') {this.value = 'DeviceHive API Url'; this.style.color = '#777';}"><br>
      <input type="text" name="deviceid" style="color: #777;" value="DeviceId" onfocus="if (this.value == 'DeviceId') {this.value = ''; this.style.color = '#000';}" onblur="if (this.value == '') {this.value = 'DeviceId'; this.style.color = '#777';}"><br>
      <input type="password" name="password" style="color: #777;" value="AccesKey" onfocus="if (this.value == 'AccesKey') {this.value = ''; this.style.color = '#000';}" onblur="if (this.value == '') {this.value = 'AccesKey'; this.style.color = '#777';}"><br>
      <select name="sensor">
	  <option disabled>Выберите датчик:</option>
	  <option selected value="dht11" name="dht">dht11</option>
	  <option value="dht22" name="dht">dht22</option>
	  </select><br>
	  <select name="gpio">
	  <option disabled>Выберите GPIO:</option>
	  <option value="gpio0">gpio0</option>
      <option selected value="gpio1">gpio1</option>
      <option value="gpio2">gpio2</option>
	  </select><br><br>
      <input type="button" value="Запуск" onClick="run(this.form);">
    </form>  
  </body>
</html>

Nikolay-Kha, 2015-09-09:
Достаточно поправить на 23 строке:
var PIN = 0;
на любой нужный пин, т.е. например для GPIO4 нужно написать:
var PIN = 4;
на той строке. Судя по коду по ссылке, вы хотите сделать формой. Тогда, форму надо поправить на:
<select name="gpio">
          <option disabled>Выберите GPIO:</option>
          <option value="0">gpio0</option>
          <option selected value="1">gpio1</option>
          <option value="2">gpio2</option>
</select>
т.е. само значение номера gpio должно быть цифрой. И еще нужно добавить одну строку в функцию run:
function run(form){ 
    print("Connecting... Refresh page for canceling.", "black");
    localStorage["dhurl"] = form.url.value;
    localStorage["dhdeviceid"] = form.deviceid.value;
    localStorage["dhpassword"] = form.password.value; 
    PIN = parseInt(form.gpio.value);  // вот эту строку
    alert(PIN); 
    dht22 = form.dht.value == "dht22";  
    get_data(data_recevied);
  }

edgiru, 2015-10-14:
Доброе время суток.
Будут ли новые релизы и что интересного нас ждет в прошивках, хотелось бы узнать планы (если это не секрет фирмы)?
извините если Вам покажется вопрос глупым. почему все на английском в devicehive да и вообще в датаарт?

Nikolay-Kha, 2015-10-14:
Здравствуйте.
Работы над релизом версии 0.4 идут. В новой версии будет поддержка новых плейграундов (мы недавно обновили сайт http://devicehive.com ) и возможно стоит ожидать появление режима энергогсбережения, но это пока не точно.
Датаарт это международная компания. Отсюда и английский язык.

closed 2015-11-05 by Nikolay-Kha