devicehive/esp8266-firmware #6: попробовал 0.4 версию

edgiru, 2015-10-31:
не хочет ни как регистрировать device не появляется в админке. хотя все так же делаю как и с прошлой прошивкой.

Nikolay-Kha, 2015-11-01:
Ключ из админки новый брали или как в прошлой версии случайный набрали? Теперь ключ, надо брать из админки и он называется AccessKey вместо DeviceKey.
Устройства с тем же DeviceId не создавали ранее? Удалите его в админке, если да. Попробуйте поменять DeviceId на другой, т.к. возможно кто-то уже создал устроство с таким id (справедливо только для плейграундов - там общее пространство имен девайсов для всех пользователей)

edgiru, 2015-11-02:
configure
1. wifi
2. pass for wifi
3. API url
4. DeviceID
5. Access Key

Делал все так. Не получается. К wifi подключается, но к админке не хочет. модуль обычный ESP01. Пробовал в админке генерировать новый Access Key, не помогло. Пробовал делать разные DeviceID которых точно не должно быть в базе, тоже не помогло.

Nikolay-Kha, 2015-11-03:
У созданных ключей в админке если кликнуть на details что внутри? Если "No permissions set" то ничего удивительного. Нажмите на ключе "edit", "add new permission" и ничего не трогая "Save". В результате в "details" все поля должны стать "Any". После этого ключ должен работать. Возможно глупый вопрос, но ключ точно весь копируете? Он должен быть ввида
r/WcTMHpGegejjySF1jsjtWQDlr9m4fyWLvnZfHuHrY=
т.е. без пробелов везде, с равно на конце.

edgiru, 2015-11-03:
доп. ключ созданый мной имеет все доступы,а ключ "Default access token for" который был уже в админке по умолчанию имее тт так же все доступы
Any
Any
Any
Sample network for .....
Any

edgiru, 2015-11-04:
я попал в "заподню". удалил все права у ключа который по умолчанию. теперь меня не пускает в админку.

Nikolay-Kha, 2015-11-05:
ох, это кажется баг в плейграундах. Я передам о нём. И вам доступ починят.
Устройство удалось зарегистрировать на плейграунде пока доступ еще был?

demon-xxi, 2015-11-05:
@edgiru, скиньте пожалуйста свой email логина на playground сюда или на info@devicehive.com.
Я поправлю настройки ключа.

demon-xxi, 2015-11-05:
@edgiru, ключик поравили. пробуйте теперь.

edgiru, 2015-11-06:
Все заработало большое спасибо! И модуль зарегистрировался.

closed 2015-11-06 by edgiru