devicehive/esp8266-firmware #8: Бесконечный "Send poll request..."

ya-suhorukov, 2015-11-04:
Ждал всю ночь, но после настройки устройства в dmesg было сообщение о том, что устройство успешно зарегистрировано и подключено к playground, но дальше следует бесконечное повторение строчки "Send poll request...". Что делать?

ya-suhorukov, 2015-11-04:
[image]

Nikolay-Kha, 2015-11-05:
poll request означает что устройство успешно подключилось и посылает запрос на сервер с целью получения команды. Это нормальный рабочий цикл. У вас все работает.

ya-suhorukov, 2015-11-05:
Но устройство не появилось в devices - playground

ya-suhorukov, 2015-11-06:
Рандомные действия привели к появлению всех устройств, спасибо)

closed 2015-11-06 by ya-suhorukov