devicehive/esp8266-firmware #9: документация

edgiru, 2015-11-07:
Без документации на русском языке крайне сложно понять что и как. Подскажите планируется ли вообще перевод, если да то примерно когда?

Nikolay-Kha, 2015-11-09:
Перевода именно документации на русский не будет. Будет еще одна статья о использование прошивки на русском.
Касаемо Dashboard, это на самом деле есть не что иное как Freeboard - https://github.com/Freeboard/freeboard
причем репозитории который вы указали - всего лишь коннектор сервера к данному проекту. Сразу подключенный к самому Фрибоарду коннектор в этом репозитории - https://github.com/devicehive/freeboard
Freeboard это стороний проект, имеющий собственную документаци, мы лишь написали коннектор для нашего сервера.

edgiru, 2015-11-09:
Ох как жаль. Придется изучать и переводить словарем. (( последний вопрос именно про Ваш коннектор к freeboard.

[image]
При выборе плагина DeviceHive в datasources в поле "Device IDS" я так понял забивать Device ID моего модуля? А в Notification names что писать (я забиваю произвольное имя)? во все остальные поля данные сами автоматом проставляются такие как  key и APIurl. Но на главной странице freeboard "последнее обновление" пишется never, так и должно быть? 
[image]

Извини уж что я с этими вопросами к Вам, не знаю у кого спросить. А про документацию к freeboard спасибо буду изучать из источника оригинала.

Nikolay-Kha, 2015-11-09:
Проект коннектора для фрибоарда делался не для esp8266. Текущая реализация коннектора позволяет отображать только нотификации из сервера DH. Поэтому отображение данных от esp8266 в текущей реализации затруденено. Из интересного сейчас, можно например отображать данные с ADC на графиках. Нужно запустить измерение ADC с каким-то интервалом времени на чипе, послав комманду откуда-то извне(из той же админки например). Затем фриборд увидев наличие нотификаций сможет что-то отрисовать. В частости 'last update' должен будет начать обновляться. В notification names ничего писать не надо. Далее, нужно создать какой нибудь контрол ("add pane"), в нём создать виджет (кнопка плюсик там есть), выбрать тип, затем внутри value написать на js код который будет превращать нотификаций в числа для отсчетов. Входной объект datasource. Результатом должно быть что-то численное.
[image]
вот пример настройки гауджа - 
[image]
код - datasources["demo"]["parameters"]["0"]
во время набора кода, если нотификации были (тогда и "last updated" будет обновлен) фрибоард сам будет подсказывать объекты.

edgiru, 2015-11-09:
Спасибо огромное Николай!

closed 2015-11-09 by edgiru