devicehive/esp8266-firmware #10: Проблема с использованием adc/read, ошибка base64

Ryo-Zi, 2015-11-14:
Пытаюсь считать данные с ADC, в панели управления данные нормально отображаются.
При попытке сделать JS скрипт для работы, после успешного считывания данных с датчика (в панели управления) и передачи ответа сервера в функцию:
```javascript
function datarecevied(base64) {
    var data = atob(base64);
    print("ADS = " + data, "green");
    measure();
}
```
В консоли браузера получаю ошибку:
> Uncaught InvalidCharacterError: Failed to execute 'atob' on 'Window': The string to be decoded is not correctly encoded.

Я делаю что-то не так или данные с сервера приходят "битые"?

Nikolay-Kha, 2015-11-14:
С ADC данные приходят в json внутри которого числа с плавающей запятой напротив каждого входа ADC(у esp8266 он один).
т.е. данные в результате команды будут ввида {"0":"0.554"}. В функции которая зовет datarecevied(base64) нужно убрать извление поля "data", а вместо этого извлечь поле "0". Полученное число с плавающей запятое - напряжение на входе ADC в Вольтах. На ADC можно подавать от 0 до 1 Волта. Дискретность ADC - 10 бит.

Ryo-Zi, 2015-11-16:
Спасибо за помощь ;)

closed 2015-11-16 by Ryo-Zi