devicehive/esp8266-firmware #12: Build errors

sigalabs, 2016-01-16:
Hello, i am trying to build this on Ubuntu 14.04 x64.

i have followed the instructions from here for the crosstool-ng installation
https:/github.com/esp8266/esp8266-wiki/wiki/Toolchain

I run the make command on the -src folder and i am getting the following output.

/opt/Espressif/crosstool-NG/builds/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: cannot find -lc
collect2: error: ld returned 1 exit status
make: *** [build/devicehive.elf] Error 1

any idea? Thanks!

Nikolay-Kha, 2016-01-16:
Hello @sigalabs 
There isn't 'libc.a' in your environment. Hope you did this steps: https://github.com/esp8266/esp8266-wiki/wiki/Toolchain#installing-xtensa-libraries-and-headers. If so, just do:
cp /opt/Espressif/ESP8266_SDK/lib/libc.a <path to this firmware>/sdk/lib

sigalabs, 2016-01-16:
Thank you for your fast response!

I have done the step  "Installing Xtensa libraries and headers" but i didn't know i have to copy the libc.a file.

Problem solved! Thanks!

closed 2016-01-16 by sigalabs