devicehive/esp8266-firmware #18: Проблема с прошивкой

Umasterov, 2016-10-30:
Пытаюсь прошить ESP последней версией, но флешер не может найти порт, к которому подключена плата. При этом терминал находит порт и подключается к ESP. В чём может быть причина проблемы?

Nikolay-Kha, 2016-10-30:
Попробуйте указать порт вручную. 
./esp-flasher /dev/ttyUSB0
или для windows
esp-flasher.exe COM4
Номера портов соотвественно замените.

Nikolay-Kha, 2016-12-01:
Удалось ли прошить?

closed 2016-12-28 by Nikolay-Kha

Umasterov, 2017-01-28:
Не удалось, не находит устройство.
Можно как-то проверить работоспособность модуля ESP? При подключении питании один раз моргает синий диод и больше ничего не горит. При этом модуль сильно греется.
Да, и, скорее всего, на моём модуле уже установлена прошивка (помню, что ранее прошить его получалось)

Nikolay-Kha, 2017-01-30:
Проверить можно. Нужно подключиться к серийному порту на скорости 74880(нестандартная скорость) и наблюдая вывод перезагрузить чип. При нормальном запуске должен быть такой вывод:
[image]
Подключиться можно например программой Putty ((в ней можно задать не стандартную скорость порта)
Этот вывод дает сам чип еще до старта прошивки. Если вывода нету - то чип умерший.

Umasterov, 2017-01-30:
А какие пину и куда нужно при этом подключить?

Nikolay-Kha, 2017-01-31:
просто нужно подключить USB конвертер к плате и замыкать пин RST на земля для перезагрузки.

Umasterov, 2017-02-05:
Подключаюсь к порту через Putty. Пустой экран терминала. Подключаю RST к земле -- мигает диод на ESP, в терминале ничего не меняется. Отключаю RST от земли -- ещё раз мигает диод и опять ничего не меняется. Может дело в USB конвертере?

Nikolay-Kha, 2017-02-06:
Может. Чтобы проверить USB переходник, нужно просто замкнуть выводы Rx и Tx, запустить терминал и все что печатается в терминал должно приходить обратно, т.е. должен  быть виден набираемый текст.

Umasterov, 2017-02-06:
Подключаться через Putty на той же скорости?

Nikolay-Kha, 2017-02-06:
да

Umasterov, 2017-02-16:
У меня просто курсор в терминале перескакивает на начало строки

Nikolay-Kha, 2017-02-17:
При вводе текста  в окне терминала при замкнутых выводах RX и TX курсор перескакивает на начало строки? Я правильно понял?

Umasterov, 2017-02-17:
Да, всё верно

Nikolay-Kha, 2017-02-17:
И набираемых символов текста не видно вообще?

Umasterov, 2017-02-17:
Если замкнуть RX и TX, то видно. Замыкаю, подключаюсь, пишу в терминале:
test**I**
Нажимаю ввод, получаю:
**I**test
Т.е. при этом просто курсор на начало строки перескакивает. Новые строки не добавляются. Так и должно быть?

Nikolay-Kha, 2017-02-17:
да, такое бывает. Приложение просто посылает не верный символ возврата строки (\r вместо \n). Переходник рабочий.
Похоже что что-то с модулем, скорее всего где-то попросту нету контакта от самогой микросхемы по пинок RX или TX. Можно еще попробовать подключить пин RX переходника к пину GPIO2 (этот пин также должен дублировать данные передаваемы на TX) - возможно удасться что нибудь увидеть.