devicehive/esp8266-firmware #19: PCF8574 pins set 'low' toggle

oli4gate, 2016-11-01:
Hi, I have connected a pcf8574 to an esp8266-01 and testing 'devices/pcf8574/write'
the issue is that pins where a '0' is written to will toggle each time we send the same json string.
Pins where '1' is written to stay unchanged.

for example, if i push the JSON string 

'devices/pcf8574/write' {"0":"0","1":"1","2":"0","3":"1","4":"0","5":"1","6":"0","7":"1","SDA":"0","SCL":"2","address":"0x7E"}
push
now pins 0,2,4,6 will go 'low' ; pins 1,3,5,7 are 'high'

(copy and push the same string)

copy
'devices/pcf8574/write' {"0":"0","1":"1","2":"0","3":"1","4":"0","5":"1","6":"0","7":"1","SDA":"0","SCL":"2","address":"0x7E"}
push
now pins 0,2,4,6 will go 'high' ; pins 1,3,5,7 stay 'high'

(copy and push the same string)
copy
'devices/pcf8574/write' {"0":"0","1":"1","2":"0","3":"1","4":"0","5":"1","6":"0","7":"1","SDA":"0","SCL":"2","address":"0x7E"}
push
now pins 0,2,4,6 will go 'low' again ; pins 1,3,5,7 stay 'high'

so the pins that I write 0 to will toggle each time.

Nikolay-Kha, 2016-11-02:
Hi! Thank you for reporting this issue. We added pcf8574 support mostly because of 1602 LCDs support and haven't tested it properly. Issue is fixed. Since we haven't announce about public release yet, I just reuploaded v0.5 release here - https://github.com/devicehive/esp8266-firmware/releases/tag/v0.5
Download it and flash chip again. On chip settings will be saved.

oli4gate, 2016-11-02:
Thank you Nikolay, for your quick reply/solution.
I have downloaded and flashed the new devicehive.bin file to the device.
After testing it I can confirm that it now works fine, issue is solved!

Nikolay-Kha, 2016-11-02:
Thanks for confirmation

closed 2016-11-02 by Nikolay-Kha