devicehive/esp8266-firmware #20: no answer from divece

Doeran, 2016-11-10:
Привет
пытаюсь прошить esp-12

контроллер определяеться ии начинаеться прошивка но падает на шаге:
Writing block 342/344
no answer from device

вывод в консоль с ошибкой по ссылке: 

http://joxi.ru/82QYLZbH1YJjz2

Подскажите в чем проблема? 
Спасибо.

Nikolay-Kha, 2016-11-10:
Привет. Модуль точно имеет 512 или более КБайт памяти?
Определить объем можно установив утилиту - https://github.com/espressif/esptool
и выполнив `esptool.py -p COM3 flash_id`
ответ будет примерно такой
```
Connecting...
Manufacturer: e0
Device: 4016
```
device это и есть id памяти. Список возможных можно найти здесь - https://code.coreboot.org/p/flashrom/source/tree/HEAD/trunk/flashchips.h
В примеры выше это чип A25LQ032 у которого 4МБайта (32 МБита) памяти.

так же можно попробовать прошить прошивку с помощью это же утилиты:
`esptool.py -p COM4 write_flash 0x0 devicehive.bin`

Doeran, 2016-11-10:
сделал
вот ответ
http://joxi.ru/YmEn3KbIZdOZw2

Nikolay-Kha, 2016-11-10:
да, у модуля 512 Кбайт.
Очень похоже на аппаратную проблему. Где в районе 352256 байта он не прошивается. Попробуйте использовать другой модуль.

Еще можно попробовать зашить прошивку побольше, хотя шансы что это изменит крайне малы. Зашейти в него нули на всю память. Файл с нулями в атаче
[512k_of_zeros.zip](https://github.com/devicehive/esp8266-firmware/files/583468/512k_of_zeros.zip)
```
esp-flasher 0x0 512k_of_zeros.bin
```
Затем запустите `esp-flasher` как делали в самом начале (без параметров из папки прошивки, он должен дефолтный конфиг еще в чистый чип записать). Если не получится, то чип можно будет использовать только для прошивок объемом 352256 байт или менее. Ну или пробовать менять саму микросхему A25LQ032.

sharptek, 2016-11-12:
Добрый день. У меня та же проблема. Система Windows 7x64. Отваливается прошивка на 342 блоке из 344. 
Нулями прошивается нормально. Все 512КБ.

Мне кажется это из-за неверного размера прошивки в релизе https://github.com/devicehive/esp8266-firmware/releases/tag/v0.5
Вот скрин сколько весит прошивка, и сколько она занимает на диске.
http://joxi.ru/v29WGLvFGVVwMr
После этого я добил прошивку NULL-ами до 352256 байт.  http://joxi.ru/L210G9Nu655jLm
Перепрошил. Прошивка залилась, но на заливке дефолтных конфигов отвалилось.
http://joxi.ru/EA4k4QgUDLL672

Перепоршил коммандой
esp-flasher 0x0 devicehive.bin
прошилось без стандартных настроек. Но зайти в терминал после перезагрузки не могу
http://joxi.ru/1A5zG5MhKBB6Jr

Nikolay-Kha, 2016-11-14:
Объём файла занимаемый на диске зависит от размера кластера(помоему это так называется в терминах ОС WIndows, вообще block size или allocation unit) файловой системы на диске. Так что это немножко о другом.
А вот то, что такое поведение проявляется на нескольких чипах - это уже интересно.
Попробуйте зашить этот релиз - 
[devicehive.zip](https://github.com/devicehive/esp8266-firmware/files/588914/devicehive.zip)
```
esp-flasher 0x0 devicehive.bin
```
В этом файле прошивка v0.5 вместе с дефолтным конфигом и все выравнено до 512КБ.

sharptek, 2016-11-14:
Да. Спасибо. Всё отлично прошилось. Никаких ошибок. 

Правда, я могу зайти в терминал. 
Прошил. Отключил от USB, отключил питание. Потом подключил к компьютеру. Терминал почему-то не определяет девайс.
http://joxi.ru/EA4k4QgUDboB82
Девайс как и раньше определяется. Все нормально. Может при подключении к терминалу нужно задать какой-то ключ. (Там подсказка, если номер порта больше 20, но у меня COM3)
http://joxi.ru/Dr86PdDukNog32

Nikolay-Kha, 2016-11-14:
Автоопределение терминала работает так: посылается байт равный 0x03 (в терминале это эквивалентно нажатию Ctrl+C) и терминал ждет ответ '^C\r\n$ '. Если  такой ответи не приходит - значит терминала там нет.
Можно руками указать номер порта для терминала.
```esp-terminal COM3```
терминал там врядли появится, но можно попытаться увидеть причину того что происходит. Подключившись таким образом к терминалу, перезагрузите чип, не отключая порта (отключив питания от самого модуля или притянув кратковременно ножку RST к земле). Будет ли какой нибудь вывод в терминале?

Nikolay-Kha, 2016-11-14:
И кстати переходники USB->TTL вы какие используете? На основе чипа FT232?

sharptek, 2016-11-14:
Использую вот такие девайсы:
http://joxi.ru/Dr86PdDukNzZy2
Переходник на основе чипа FTDI 232
http://joxi.ru/krDORLbT0RJMeA
http://joxi.ru/82QYLZbH1ly182

После того как встал на прослушку порта и перегрузил контроллер (через RST на землю) на консоли такой странный набор символов:)
http://joxi.ru/823kv9qU6ZDJG2
А вот это в дефолтной консоли Windows
http://joxi.ru/Vm6Q46Gtxyj40r
Может не та кодировка консолей? Или так и должно быть? 

P.S. Спасибо большое за помощь, что не бросаете))

Nikolay-Kha, 2016-11-14:
Дело не в кодировке, а в скорости порта.  Если бы прошивка стартанула бы, то порт был бы быстро переключен в нужную скорость и вывод был бы читаем. Она там видимо вообще не стартует.

Но теперь мне кажется причина больше в переходнике. [Здесь](https://geektimes.ru/post/258842/) статья которая рассказывает что может творить фирменный драйвер, в комментариях еще апдейты есть с появлением новых версий. Чудес происходящих с FT232 возможно очень много. Есть ли возможность поменять переходник? С CP2101 и CH340 подобных проблем никогда не происходит. Или прошить из под операционной системы Linux? (там драйвер работает всегда)
Еще можно попытаться прошить на заниженной скорости. В архиве - 
[esp-flasher-19200.zip](https://github.com/devicehive/esp8266-firmware/files/589411/esp-flasher-19200.zip) - прошивальщик который работает на скорости 19200. Попробуйте прошить им. 

p.s. это вам спасибо, что просто не бросили попытки прошить)))

closed 2016-12-28 by Nikolay-Kha