devicehive/esp8266-firmware #22: build errors

MaxLesichniy, 2016-11-27:
Не получить собрать прошивку в eclipse. С чем могут быть связаны эти ошибки ?
[image]
[image]

Nikolay-Kha, 2016-11-27:
Стандарт языка Си видимо не тот. В gcc версии 4.х по умолчанию использовался gnu89. Начиная с пятой версии по умолчанию используется gnu11. Видимо у вас gcc 5 или более новой версии. Оригинальный SDK использует фичи, которые требуют стандарта gnu89. Так что, надо просто к CFLAGS добавить -std=gnu89 . Я явно указал это в последнем коммите бранча develop. Но можете по примеру этого коммита - https://github.com/devicehive/esp8266-firmware/commit/d47a4f448726f0fbadd17c3bf54e103ca959d71b добавить параметр к коду любой версии. Должно собраться.

MaxLesichniy, 2016-11-27:
Все заработало, спасибо)

closed 2016-11-27 by Nikolay-Kha