devicehive/esp8266-firmware #23: Быстродействие

MaxLesichniy, 2016-11-30:
Здраствуйте, Николай. Подскажите пожалуйста, что может быть. Использую прошивку esp v0.5. и интегрировал ios framework в свое приложение. Управляю для теста светодиодом.  Возникают задержки, причем разные, устройство может сразу обработать или только при следующем poll. Может быть проблема с моей стороны? Или это из-за http и нужно переходить на sockets. Есть ли поддержка сокетов в прошивке esp?

Nikolay-Kha, 2016-11-30:
Здравствуйте! Уточните, пожалуйста, как вы обращаетесь к esp8266. Через сервер DeviceHive или используя локальный REST на чипе? Если через сервер - то где он расположен относительно чипа?

MaxLesichniy, 2016-12-01:
Через dashboard

Nikolay-Kha, 2016-12-01:
dashboard который на плейграунде тут - http://playground.devicehive.com/dashboard/ ?

MaxLesichniy, 2016-12-01:
извините, admin panel на плейграунде

Nikolay-Kha, 2016-12-01:
Ага, т.е. получается вы используете плейграунды. На них добиться работы в рилтайме не получится - они расположены далеко и нагрузка на них не предсказуема. Если нужна более стабильное рил тайм применение - подниме свой сервер. Это легко:
Поставьте докер ( https://docs.docker.com/engine/installation/ ) на линуксовую или OS X машину (например на свой ноутбук), затем запустите:
docker run -p 80:80 devicehive/devicehive-standalone
через пару минут поднимется сервер с админской панелью на http://айпи_или_домен_куда_ставили/admin 
Логин/пароль -  dhadmin/dhadmin_#911
Сразу зайдите в админке в Users -> dhadmin -> networks -> веберите"Virtual Led sample network" и нажмите на "grant access". После этого можете создавать ключи доступа, подключать устройства итд.

closed 2016-12-28 by Nikolay-Kha