devicehive/esp8266-firmware #30: Переподключение к серверу

reacct, 2017-02-02:
Добрый день!
Подскажите, пожалуйста, про процесс переподключения к серверу.
При отключении интернета (wifi включён), устройство либо повисает, либо выдаёт "ip:[свой ip],mask[маска],gw[ip точки]" (смотрю через dmesg). При включении интернета ситуация не меняется. Указанную строчку в прошивке пока не нашёл. Помогает только перезагрузка устройства.

Например, в других случаях, 
1. когда пропадает и появляется wifi, 
2. когда пропадает wifi, появляется без интернета, а потом появляется интернет,
        устройство подключается в подавляющем большинстве случаев.

Версия 0.5

Nikolay-Kha, 2017-02-03:
Здравствуйте,
В моменты подключения и отключения происходи вызов этого колбэка - https://github.com/devicehive/esp8266-firmware/blob/develop/firmware-src/sources/dhconnector.c#L346
В случае, когда интернета нет, но вайфай подключе, прошивка должна пытаться подключиться к серверу все время и происходить вызов этого колбэка - https://github.com/devicehive/esp8266-firmware/blob/develop/firmware-src/sources/dhconnector.c#L51 (ошибка при попытке подключиться/послать что нибудь по сети)
Вывод о котором Вы пишите, идет не из самого кода прошивки, а из SDK чипа.

Вообще, выглядит как баг, я попробую повторить.

Nikolay-Kha, 2017-02-10:
проверил - да, это баг.  Исправим.

Nikolay-Kha, 2017-02-12:
Исправленно. Спасибо за репорт. Можете собрать из исходников или дождаться релиза новой версии в ближайшее время.

closed 2017-03-10 by Nikolay-Kha