devicehive/esp8266-firmware #39: Работа I2C

reacct, 2017-10-18:
Добрый день, Николай!
Решил попробовать подключить память 24c04 через I2C, и что-нибудь записать, а потом считать. Ни через playground, ни напрямую, используя функции (записать любое слово, а потом вывести его в uart), не получается сделать запись или чтение (может быть что-то одно). Чтобы исключить неработоспособность модуля памяти, пробовал ещё один экземпляр - ситуация та же. Под управлением библиотек arduino ide, esp пишет и читает через I2C у 24c04, т.е. модули рабочие.
Видимо, делаю что-то не так.
Подскажите, пожалуйста, порядок работы с I2C.

Nikolay-Kha, 2017-10-20:
Здравствуйте.
чтение/запись по i2c проста - две команды i2c/master/read и i2c/master/write. Указываете адрес и байты которые читать/писать. Все данные кодируются с помощью base64.
Хотелось бы увидеть какие запросы вы посылали в 24c04, сами поля параметров команды и что приходило вам в ответ. Без этих сведений гадать сложно.
На вскидку, возможно i2c адрес указан не правильно.  На встроенном веб сервере есть сканер адресов шины i2c - scani2c.html. Запустите его и увидите по каким адресам отвечает 24c04, используйте их в командах.

reacct, 2017-10-20:
Спасибо! Буду пробовать дальше. Можно ещё уточняющие вопросы?
Почему в команде на чтение необходимо указывать адрес в шестнадцатеричном формате, а на запись в десятичном?
При указании количества байт на чтение, речь идёт о количестве байт исходной строки или закодированной в base64?

Nikolay-Kha, 2017-10-20:
В документации ошибка. Код считывающий адрес в обоих случаях работает один и тот же. Поправил доки. В обоих случаях адрес должен быть шестнадцатеричным (`7A` например, можно с префиксом `0x` - `0x7A`).
Количество байт на чтение - исходное количество байт, передаваемых по i2c.

reacct, 2017-11-06:
Добрый день!
Подключиться получилось. Действительно, адреса были не те. scani2c.html помог. У 24c04 два адреса, у 24lc04b – 8. Причём с [адресом+1] также можно работать у обоих моделей. Однако дальше стало интереснее. И вопросов возникло ещё больше.

Запись данных происходит ограничено: из любого количества байт на запись (<=264 разумеется) записываются только с 1го по 15й байты, т.е. без 0го и 16-264.
Чтение идёт со смещением данных: если вся память заполнена полезными данными, то при каждой следующей команде идёт смещение полученных данных на 8 байт (если не ошибаюсь. Либо 16, если напутал), т.е. начало полезных данных пропадает, а в конце записываются FF.

Тестировал всё с помощью API на внутреннем веб-сервере, также использовал ещё одну ESP со скетчем на ардуино для контрольных записи/чтения. Память 24с04 и 24lc04b.

Как исправить эти две проблемы? / Что я делаю не так?
Можно ли писать в определённые байты?
Размер на запись/чтение ограничивается какими-то особенностями кода или только переменной в конфиге?
Для чего нужно поле "data" при чтении?

Nikolay-Kha, 2017-11-07:
> записываются только с 1го по 15й байты, т.е. без 0го и 16-264.
а вы адрес страницы первым байтом в пакете передаете? по документации на память - http://www.atmel.com/images/doc0180.pdf первым байтом нужно указывать word address. При чтении, сначала нужно записать этот адрес, а потом читать.

> Как исправить эти две проблемы? / Что я делаю не так?
> Можно ли писать в определённые байты?
см. выше.

> Размер на запись/чтение ограничивается какими-то особенностями кода или только переменной в конфиге?
Размер ограничен размерами буфера в коде. Объявлен как INTERFACES_BUF_SIZE в user_config.h.  264 байта сделано специально, что бы поместились 256 байт данных + пару байт на заголовки типа установки адреса у данного чипа.

> Для чего нужно поле "data" при чтении?
Сюда можно помещать данные которые будут записаны в чип перед чтением, например чтобы указать чипа что именно мы хотим прочитать. В случае данного чипа памяти сюда можно как раз класть word address и передача будет идти как на "Figure 11. Random Read" в документе выше, т.е. сначала запись адреса, затем старт, затем запись адреса чтения и собственно чтение.

reacct, 2017-11-08:
Действительно, про word address позабыл. Чтение и запись происходят нормально. Буду дальше тестировать.
Благодарю за помощь!

closed 2017-11-08 by reacct