devicehive/esp8266-firmware #40: использование platform.io

team55, 2017-11-06:
Не смотрели ли в сторону среды platform.io?
Попытался портировать проект под нее, пока не удачно

Использовал devicehive.ld из проекта + исходники, подправил platformio.ini, прописал параметры
elf собирается в чип заливается - не работает (в терминал мусор летит)
конвертировал elf в bin средствами  genbin.sh
залил с помощью  esp-flasher, на этапе заливки юзерского конфига зависает

что я упустил, на что обратить внимание?

team55, 2017-11-06:
хм, версию 0.2 удалось собрать..

team55, 2017-11-06:
[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
build_flags = -Wl,-T"eagle.app.v6.ld"

Nikolay-Kha, 2017-11-07:
> Не смотрели ли в сторону среды platform.io?

нет

> elf собирается в чип заливается - не работает (в терминал мусор летит)

 использовали ли Makefile из проекта?

> что я упустил, на что обратить внимание?

на ошибки в выводе Makefile, скриптов и результат их выполнения(exit status процесса). Возможно еще где-то переполнилась какая нибудь область памяти, хотя в этом случае скрипты бы упали с ошибками, но может где ошибка в скриптах.

> хм, версию 0.2 удалось собрать..

там память чуточку иначе была организована.

> [env:nodemcuv2]

я честно говоря даже не понимаю откуда этот конфиг...

team55, 2017-11-07:
конфиг от этой штуки http://platformio.org/ 
в ней можно собирать проекты для esp8266, как с использованием фреймворков (arduino, simba, rtos) так и голые (sdk идет в комплекте)
Версию 0.2 собрал
Для 0.7 использовал для сборки параметр
build_flags = -Wl,-T"devicehive.ld"
pages.h сгенерил пока вручную

похоже пока проблема в двух файлах и дефолтных настройках которые заливаются прошивальщиком. На выходных попробую по шагам пройтись.