devicehive/esp8266-firmware #42: Authorization header check only in one variant

team55, 2017-11-28:
модуль httpd.c
В rest запросах заголовок Authorization проверяется только на вариант 'Authorization', тогда как многие клиенты преобразуют его в нижний регистр. 
К примеру так делает react-native при работе через библиотеку okhttp, и на это нельзя повлиять..
Разный регистр заголовков вполне соответствует стандарту
https://www.w3.org/Protocols/rfc2616/rfc2616-sec4.html#sec4.2

Логичнее было бы и проверять заголовки без привязки к регистру (либо преобразовать верхний-нижний)

team55, 2017-11-28:
https://github.com/facebook/react-native/issues/16988 думал проблема в ReactNative

Nikolay-Kha, 2017-11-29:
Да... по спекам хедеры могут быть в любом регистре. Поправим. Спасибо.

Nikolay-Kha, 2017-12-04:
Исправлено. Дайте знать если нужна бинарная сборка.