devicehive/esp8266-firmware #43: Поддержка bme280

aver-ua, 2018-02-07:
Планируется ли поддержка датчика bme280?

Nikolay-Kha, 2018-02-08:
хм, действительно стоит ее как нибудь сделать. По срокам сказать затрудняюсь.
BMP280 отличается от BME280 наличием гигрометр. Текущая команда  https://github.com/devicehive/esp8266-firmware/blob/develop/DeviceHiveESP8266.md#devicesbmp280read может считать данные и с BME280, однако без влажности.

aver-ua, 2018-02-08:
Спасибо, понятно. 
Т.е. BME280 должен работать. Ок, попробую, когда датчик придёт

username-111, 2019-01-19:
Подскажите пожалуйста, где взять бинарный образ прошивки с поддержкой этого датчика? А то последний доступный от 23.11.2017 т.е. старше вашего коммита.

Nikolay-Kha, 2019-01-21:
собрал специально для вас. Только бинарный файл прошивки, утилиты и прочее берите с последнего релиза если нужны.
[devicehive.bin.zip](https://github.com/devicehive/esp8266-firmware/files/2778550/devicehive.bin.zip)

username-111, 2019-01-26:
Значение влажности почему-то всегда приходит 0. Проверял на другой [прошивке](http://embedded-lab.com/blog/making-simple-weather-web-server-using-esp8266-bme280/) - датчик исправен, влажность измеряет.