devicehive/esp8266-firmware #44: NodeMCU v3 Lolin esp8266mod Hysiry нет подключения

phsin, 2018-02-11:
после прошивки терминал не подключается к esp.
Подскажите, пожалуйста, как можно исправить?

Nikolay-Kha, 2018-02-12:
попробуйте запустить 
esp-terminal-win.exe com10
т.е. указав порт вручную и перезагрузить саму плату. Будет ли вывод какой либо в терминале?

-> порт не отвечает, не удается конфигурировать устройство
не вижу подобного вывода на скриншотах. Если где-то подобные тексты проскакивают, значит возможен порт занят чем-то в системе - надо попробовать перезагрузить сам компьютер.

phsin, 2018-02-13:
> попробуйте запустить
>esp-terminal-win.exe com10
>т.е. указав порт вручную и перезагрузить саму плату. Будет ли вывод какой либо в терминале?

подключается к com10, при перезагрузке платы выводятся не читаемые символы
[image]

phsin, 2018-02-13:
http://forum.espruino.com/conversations/305323/ предлагают изменить адрес загрузки и частоту, 
как это сделать?

Nikolay-Kha, 2018-02-13:
Судя по выводу - прошивки в чипе почему-то нету, хотя с NodeMCU v3 Lolin модулями обычно все работает. esp-flasher к сожалению не умеет менять частоту из параметров. Адрес загрузки менять нельзя - прошивка просто гарантированно не запуститься. Попробуйте прошить с помощью утилиты esptool ( https://github.com/espressif/esptool )
esptool.py --port <serial-port-of-ESP8266> -b 115200 write_flash 0x00000 devicehive.bin
аргумент -b задает частоту, попробуйте 115200, 74880 или на худой конец 19200.

phsin, 2018-03-20:
Прошить получилось только с внешним источником питания на 3.3В минимум на 200мА, тот что на плате на 100мА. После прошивки флешером https://github.com/nodemcu/nodemcu-flasher плата стала отвечать в мониторе который идет с прошивкой.

[image]

closed 2018-03-20 by phsin