devicehive/esp8266-firmware #48: esp-terminal can not detect port

yccy1230, 2018-03-09:
I downloaded the v0.8 devicehive-esp8266-firmware and Successfully flash the firmware with the esp-flasher. However, after the GPIO0 is unplugged and restarted the device , the device cannot be found using the esp-terminal tool.
The hardware I use is esp8266-01. Expect your help.
[image]

Nikolay-Kha, 2018-03-12:
Hello. What is the EEPROM chip on your module? That is a tiny 8pin chip nearby esp8266 itself. What is the label on it?
For example, I have module which works with 
25Q40BV
chip - that is 4MBit(512KByte) flash. This size is suitable for the firmware. The firmware is not working with less memory, and chip(and flasher as a result) doesn't return error on flashing.

And is there any output with `esp-terminal COM6(change to your port)` on chip reset?

yccy1230, 2018-03-28:
Hi. I'm glad to receive your reply.
The EEPROM chip's label is "MD BX1642 MD25D80SIG P13472"
I check my chip's flash size.It is 8Mbits( i think it may be enough),just as the follow pic shows.
And there id no output with `esp-terminal COM6(my own port)` on chip reset.
[image]

closed 2018-03-28 by yccy1230