devicehive/esp8266-firmware #49: Не работает прошивка

StDeniks, 2018-03-17:
Подключил и прошил как описано в pdf (в архиве с тулами и прошивкой), снял GPIO0 с нуля, запустил терминал, перезагрузил через ресет, в результате в терминал выкидывает квакозябры.
Прошиваю через CH340, чип ESP-12-F QIO L4.
Что я делаю не так?
[image]
Пробовал подключаться через ардуиновский терминал и перебрал в нем все частоты, на всех квакозябры.

StDeniks, 2018-03-17:
I connected and flashed esp as wrote in .pdf, then disconnected GPIO0, opened terminal and restart esp.
Result you see in the screen on the top.
I use CH340 for flash. Why it don't work?

Nikolay-Kha, 2018-03-19:
Try to flash it with esptool ( https://github.com/espressif/esptool )
$ esptool.py -p COM10 -b 115200 --no-stub write_flash -fm qio --flash_size 512KB 0x0 devicehive.bin
then check with esp-terminal
$ esp-terminal-win.exe COM10
if there is no respond, make a dump of firmware inside chip:
$ esptool.py -p COM10 -b 115200 read_flash 0x0 413307 dump.bin
and compare dump.bin and devicehive.bin files.
$ diff dump.bin devicehive.bin
if there is no difference, this command should output nothing.