devicehive/esp8266-firmware #53: Host watchdog

VladVons, 2018-05-17:
Thanks for a good job!

I use your firmware, but dont use DeviceHive internet service at all. 
ESP8266 device gets IP address via DHCP on Microtik router and Raspberry is asking sensors via your API every 2-5 seconds.
After 5-12 hours of activity some devices become inaccessible and must be rebooted with power of/on. After that they run normally again 5-12 hours and  disappear and so on. 
It would be GOOD to add to your firmware host checking watchdog option.

2 fileds in setup:
- host to check
- max fails to reboot (default 3)

thanks

evg26, 2018-10-26:
After 5-12 hours of activity some devices become inaccessible and must be rebooted with power of/on. After that they run normally again 5-12 hours and disappear and so on.

Nikolay-Kha, 2018-11-28:
Hm... I have several esp8266 which works for a few years and there is no reboots. But in DeviceHive server mode.
Can you check if esp8266 is overheating? Can you collect logs when it hangs?

evg26, 2018-11-29:
пока нет возможности проверить как проверю подскажу. не грелась, пробовал 2 раза устанавливать оба раза зависали

>Среда, 28 ноября 2018, 22:48 +06:00 от Nikolay Khabarov <notifications@github.com>:
>
>Хм ... У меня есть несколько esp8266, который работает в течение нескольких лет, и нет перезагрузки.  Но в режиме сервера DeviceHive. 
>Можете ли вы проверить, перегревается ли esp8266?  Можете ли вы собирать журналы, когда они зависают?
>- 
>Вы получаете это, потому что вы прокомментировали. 
>Ответьте на это письмо прямо,  посмотрите на GitHub или  отключите тему .

С уважением
Евгений